python-microscope/microscope #177: Remove camera transforms and data processing

dstoychev, 2021-01-06:
I think data processing is outside of the scope of python-microscope. Currently, the `Camera` abc includes the `_process_data` method, which flips and rotates the data according to the `_transform` attribute. This attribute is configured by the "transform" setting and also internally by the pvcam implementation via the `_set_readout_transform` method. I believe both can safely be removed, with the pvcam stuff seemingly being unnecessary and maybe even obsolete.

If you agree, I can provide a PR for the necessary changes, as well as a PR for Cockpit to add the ability for post-processing of image data.

mickp, 2021-01-06:
The motivation for this was so that cameras can give the same view of a region regardless of configuration changes - e.g. changing the readout mode can flip an image. Not all cameras implement transforms in hardware or their libraries, so the functionality is included as part of microscope. Where a camera does implement transforms, `_process_data` can simply be a passthrough.

iandobbie, 2021-01-06:
dstoychev <notifications@github.com> writes:

> I think data processing is outside of the scope of python-microscope. Currently, the Camera abc
> includes the _process_data method, which flips and rotates the data according to the _transform [...]

No there was a fundamental reason for having a camera rotation in
microscope, If you have 50 cameras on 5 machines, do you really want to
force the master cockpit machines to deal with all those rotations?

Also most EMCCD cameras return images with different orientations in
conventional and EM mode. As a function of the device microscope can
know about this and deal with it, putting this in cockpit is wrong.

However we also still need orations in cockpit as different beam paths
might produce images at different rotations. I know this is code
duplication but not much and I think there are cases where both are the
right place to do it.

Ian

dstoychev, 2021-01-06:
Thanks to both of you for the feedback. It seems post-processing is needed to some degree in python-microscope. I will close the issue here for now and focus on post-processing in Cockpit.

closed 2021-01-06 by dstoychev