python-microscope/microscope #188: Digitally triggered stage steps.

iandobbie, 2021-02-12:
I have added support for a digitally triggered stack along a given stage axes. The code is in https://github.com/iandobbie/microscope/tree/stage-digitial-stack

There are two components this this implementation so far.

1) an addition abc method for a stage axes setupDigitalStack which unless overridden returns 0, meaning not implemented.

2) Addition of a setupDigitalStack to the simulated stage implementation that returns 1 to indicate it is implemented. 

Still to implement a real version on the zaber Z drive to ensure that it actually works.

iandobbie, 2021-02-15:
I have implement this in the simulated stages in my branch https://github.com/iandobbie/microscope/tree/stage-digitial-stack which goes along with the cockpit branch https://github.com/iandobbie/cockpit/tree/digital-zstack

iandobbie, 2021-04-21:
A proper implementation of this is in my branch zaber-digital-stack, but seems to have got ilan's icon in there too. This closes #175 as well

carandraug, 2021-05-25:
I have looked at this implementation and I think the logic is a bit higher level for the low level control that we are aiming. I think that what we want is basically a method that enables us to place the stage in a state that it will respond to a digital signal to move by a certain fixed distance. The implementation also adds logic about number of triggers, start position, and whether it is supported via a return value.

1. Number of triggers: I don't think this is required and adds an arbitrary limitation. If we add it, we should also define what will happen after the axis has received the given number of triggers. From the only concrete implementation, it seems that the axis would continue moving even after reaching the given number of triggers, so maybe just remove it?

2. Start position: this is basically just calling `move_to` at the start of `setupDigitalStack`. The caller might not want to make any move at all, or it might want to make a relative move. I think this could be removed and we can make the caller responsible by placing the stage at the position they want. 

3. return value of zero means calling the function failed: maybe just return an exception? This prevents the case where the caller forgets to check the return value and ends up in a weird state.

Could the stage instead inherit from `TriggerTargetMixin`? This would require addition of a new method that defines what to do in receipt of a trigger (something like `set_trigger_move`). It could then work like so:

```
axis.set_trigger_move(some_distance)  # this method does not exist
axis.set_trigger(TriggerType.SOFTWARE, TriggerMode.ONCE)
axis.trigger()  # causes axis to move

# or alternatively, with hardware triggers (setting the trigger will fail
# if the stage does not support hardware triggers)
axis.set_trigger(TriggerType.HIGH, TriggerMode.ONCE)
# stage would move in receipt of hardware trigger
```