python-microscope/microscope #189: Lumencor Light Engine TCP support

dstoychev, 2021-02-18:
This one is a really low-hanging fruit, as pyserial already allows to use TCP. Besides a small change to how TCP commands are sent and received (no termination required), the main change is that instead of using the `serial.Serial` class directly, instances should be created with the `serial.serial_for_url()` function instead. This way, the user can specify any protocol and it will just work. For TCP, instead of COM port or device path, a URL needs to be used: `socket://192.168.201.200:8095`.

carandraug, 2021-04-19:
The use case behind doing this the user is already using the ethernet interface to have the web interface.

dstoychev, 2021-08-14:
Implemented here: https://github.com/dstoychev/microscope/tree/lumencor-le.