python-microscope/microscope #191: Mac log files are put in the wrong place.

iandobbie, 2021-02-24:
The log files on the mac go in the current dir which is not the expected place and breaks if the system is running in a sandbox from a double clickable icon. I have a suggested fix in my branch mac-log-dir

carandraug, 2021-02-24:
Why is it not the expected place? If we are changing the location of the log files to something system specific, this should be done for all platforms, not only for mac.

The proposed fix has two issues:

1. it only changes the location of the individual device server while leaving the `__MAIN__.log` file still in the current working directory.
2. it saves the log files in the cockpit logs directory but this is not cockpit

I believe the real reason behind this issue is not that it was a surprise to find the log files in "~/Library" but instead that it was preventing to work in the installer sandbox. A command line option to control the location of the logs file would also solve your problem and would also solve the current monkey patching we currently have to do on the testsuite to test the device server without creating the log files.

Since we already have logs in stderr by default, I propose that we just don't have log files unless some directory for log files is set on the command line.

iandobbie, 2021-02-24:
David Miguel Susano Pinto <notifications@github.com> writes:

> Why is it not the expected place? If we are changing the location of the log files to something system
> specific, this should be done for all platforms, not only for mac.
>
> The proposed fix has two [...]

Yes, I noticed that and was just about to fix that. 

> I believe the real reason behind this issue is not that it was a
> surprise to find the log files in "~/Library" but instead that it was
> preventing to work in the installer sandbox. A command line option

Yes the sandbox is the killer, but there is a spec for log files on mac
so maybe we should follow that by default if no dir specified.

> to control the location of the logs file would also solve your problem
> and would also solve the current monkey patching we currently have to
> do on the testsuite to test the device server without [...]

Ok. to be honest the simulation logs are probably totally useless unless
you are debugging something very strange.

Ian