python-microscope/microscope #193: pvcam: incomplete readout mode list

dstoychev, 2021-03-02:
At the moment, microscope lets you select the readout mode as a combination of speed and gain. This doesn't seem to work very well and the main issue is that not all possible combinations are listed. I think it will be better to list the speed and gain options separately. For example, on the Prime BSI, here are the possible combinations:

| Speed   | Gain                 |
| ------- | -------------------- |
| 100 MHz | CMS (12bit)          |
| 100 MHz | HDR (16bit)          |
| 200 MHz | Full well (11-bit)   |
| 200 MHz | Balanced (11-bit)    |
| 200 MHz | Sensitivity (11-bit) |

carandraug, 2021-04-19:
Why are not all possible combinations listed? Reading from the code, I see we are reading all listed in `PARAM_READOUT_PORT`:

    ro_ports = self._params[PARAM_READOUT_PORT].values

> I think it will be better to list the speed and gain options separately.

The current interface really expects this to be a single string. Such change would require changes on all other cameras and in Cockpit.

dstoychev, 2021-08-14:
I discovered that the problem is that readout hierarchies consist of three levels -- port, speed, and gain -- but we only check the first two. Fixed in https://github.com/dstoychev/microscope/commit/1feb84f99bd54d19a01b71c2dc372b60310e1f5b.