python-microscope/microscope #195: Harmonise the enable/disable interface

dstoychev, 2021-03-02:
I feel like disable() and enable() have too broad use at the moment. Some devices (e.g. `DeformableMirror`) take it as "prepare the device for operation", whereas others (e.g. `LightSource`) directly operate the device. I think this is a huge discrepancy in usage and I propose we stick to the former definition. As such, light sources will need a new pair of methods, something along the lines of `switch_on()` and `switch_off()`. These new methods will not be able to guarantee that the device will switch to the desired state because often hardware (TTL) signals override the software control. Example signatures:

```python
@abc.abstractmethod
def switch_on(self): -> bool:
    # Return true on successful communication, not if the device has switched on
    pass

@abc.abstractmethod
def switch_off(self): -> bool:
    # Return true on successful communication, not if the device has switched off
    pass
```

This is related to #171.

mickp, 2021-03-02:
With continuous wave light sources, turning them in *is* preparing them for operation. Generally, switching untriggered light sources on requires a settling time until the output power is stable. Modulation is achieved by some other device, such as a shutter or an AOM.  I think the proposed methods are redundant: enabling a CW light source will involve calling `switch_on`. If you call `switch_off`, it's then no longer ready for use in a measurement, so should not be considered enabled.

dstoychev, 2021-03-18:
I agree. I raised this issue while working on an implementation of a compound light source, such as the ones from CoolLED and Lumencor, where the device may be enabled yet not emitting light because it's waiting for a hardware trigger to modulate it. But it turns out that such light sources can still be implemented with the current `LightSource` class, so this issue is not longer relevant.

closed 2021-03-18 by dstoychev