python-microscope/microscope #196: Add support to PI piezo stages

gjover, 2021-03-12:
Hi,
At ALBA we are installing a CryoSIM system.
For this project we need devices for different hardware.
One of them is a z-stage from PI (physikinstrumente).

I have develop an stage device for the e754 controller, but can be generalized to other ones.

gjover, 2021-03-22:
Hi,
I just added support for rfc2217 protocol.

For serial controllers one could use ser2net to allow socket communication.
The configuration would be:
/etc/ser2net.conf
3000:telnet:600:/dev/ttyS0:19200 8DATABITS NONE 1STOPBIT

And one could construct the device as:
    
PIM687_CONF = dict(url='rfc2217://127.0.0.3:3000')
DEVICES = [
    device(pi_stage.PIm687, host, port, conf=PIM687_CONF),
]

iandobbie, 2021-03-30:
I don't understand what this provides that a wrapper using pyserial wouldn't. It  adds another dependency and a new different way to talk to devices. Unless there is a very clear advantage I think it would be better to talk to the stage with pyserial as we do with every other stage.

carandraug, 2023-01-06:
Issue #259 (now closed as duplicate of this one) suggests using PIPython (https://github.com/PI-PhysikInstrumente/PIPython) to do this.