python-microscope/microscope #198: Add support to Meadowlark SLM

gjover, 2021-03-22:
We need a microscope device to control a Midowlark SLM.

May be it would be worth to define a generic class LightModulator for SLM, rotors or atenuators.

juliomateoslangerak, 2021-08-16:
Just to mention that I have a fork from @mickp's code for meadowlark that is compatible with the ODP version of Madowlark's SDK. quite different even with the same name.
https://github.com/juliomateoslangerak/bnsdevice

iandobbie, 2021-08-16:
When we upgraded to windows 10 on the Oxford CryoSIM setup we needed a new version of the SDK and the control card driver. The HDMI control is I think a whole different setup as it uses the video card to produce the images rather than the PCIe card that our SLM's use so looks like there could be 3 separate bits of driver code required for the Meadowlark SLMs