python-microscope/microscope #199: Add support for Meadowlark rotor

gjover, 2021-03-22:
We need a microscope device to control a Midowlark rotor D5020.

iandobbie, 2021-03-22:
We actually control these on several of our systems. We use the analog input to allow synchronous control with the other digital and analog signals. There is also a digital input but that can only switch between two preprogrammed states and the SIM needs at least 3 for single colour imaging and more for multi colour imaging. 

Do be aware that the firmware we had on our original D5020 controllers meant that the analog inputs did not work properly we had to go through several versions and we are now on a file called  D5020_104.hex, which I assume is version 104

gjover, 2021-03-22:
Thank you Ian,
Since we have an HDMI SLM, it would be difficult for us to do hardware synchronization.
As a first approach we are setting voltages by software calls.

iandobbie, 2021-03-25:
Right. The one issue with this is the experiment framework is based around hardware triggering. It works to a certain extent with software triggers, however I do worry that the stage response is too slow to make this feasible. having a 0.5 s delay on each frame is ok in the mosaic as the motion often take this long but for a Z stack this could be a deal breaker. You may have to implement a new mechanism to get a more defined response time to software calls to stage motion commands. This could help our mosaic code though as this can be slow on some systems (eg https://github.com/MicronOxford/cockpit/issues/699)