python-microscope/microscope #201: device_server: index config parameter is overwritten

dstoychev, 2021-03-25:
There is an issue with the `serve_device()` function, which sets the "index" config parameter to the `count` variable, effectively overwriting the user configuration:

https://github.com/python-microscope/microscope/blob/610b93271bbb86c0e42690a606db6a74c5988a4f/microscope/device_server.py#L444

It needs to be something along the lines of:

```python
if "index" not in dev["conf"]:
    dev["conf"]["index"] = count
```

iandobbie, 2021-03-25:
Maybe we need to store this information elsewhere as my breif reading of the code suggests that this is used by floating devices (probably the andor emccd driver) to keep track of which device is which during initialisation. The issue is that you need to open the device to get its serial number to know where to bind each device if you have more than one camera on a machine.

carandraug, 2021-04-12:
As ian mentioned, the index kwarg is required for floating devices. The whole issue with floating devices is that it's not know ahead of time which is why the device server injects it. Better treat "index" as a reserved keyword.

What were you using the index argument for? Currently, the only devices that use it are floating devices.

dstoychev, 2021-04-15:
I think I may be misunderstanding how floating devices work. If I had a system with three floating cameras, each imaging at different wavelength, how am I supposed to identify which is which? Surely I must be able to determine that the camera served at a specific port is determined to work at a specific wavelength.

carandraug, 2021-04-15:
> If I had a system with three floating cameras, each imaging at different wavelength, how am I supposed to identify which is which?  Surely I must be able to determine that the camera served at a specific port is determined to work at a specific wavelength.

No. Some SDKs do not let you determine that until later. That's the problem that floating devices tries to solve. In some SDKs each camera does get assigned an index number but that index number is not fixed, it's up to the SDK to assign it, and it might assign each camera a different index number each time the cameras are started. And while you can identify the camera by its serial number, you can only do it after initialising it.  From the docs:

    Some SDKs handling multiple devices do not allow for explicit
    selection of a specific device.  Instead, a device must be
    initialized and then queried to determine its ID.

The index numbers are only useful to identify which camera is which after the cameras have been initialised which is why it doesn't make sense for the user to set them. Instead, the user sets the serial number (or UID). The device server passes an index number (0..n) that it generates, then checks the uid of the camera for that index, and then serves it as per the device-server configuration.

dstoychev, 2021-04-15:
Thanks for the explanation! I have misunderstood how this attribute should be used and as such I will be closing this issue.

closed 2021-04-15 by dstoychev