python-microscope/microscope #204: device_server: FloatingDeviceMixin in functions and controllers

dstoychev, 2021-04-26:
At the moment there doesn't seem to be a way to indirectly use -- in a user-defined function or as a subdevice of `Controller` -- classes which inherit from `FloatingDeviceMixin`. The main issue seems to be keeping track of indices across multiple processes, but there's also the matter of UIDs and how floating devices are served. Perhaps it would be best to create some sort of shared data structure, a sort of a floating device depot, which multiple processes can read and write. To achieve complete automation of the index selection, the `FloatingDeviceMixin` (or its derivatives) class should implement the synchronisation logic, using the suggested shared data structure.

carandraug, 2021-04-28:
For a user defined function, at the moment the solution is for that function to do the assignment of each device itself. For controller, I don't understand how its subdevices can be floating devices. Do you have a specific use case for this?

dstoychev, 2021-04-28:
Floating devices seem quite popular among cameras, so if a controller has one as a subdevice then this is an easy way for it to have a floating device too. Theoretical example: PR #133.