python-microscope/microscope #208: Laser power not set at startup.

iandobbie, 2021-05-25:
On DeepSIM the first time you access the 561 you get 

Exception in thread _updater:
Traceback (most recent call last):
  File "c:\python37\lib\threading.py", line 917, in _bootstrap_inner
    self.run()
  File "c:\python37\lib\threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "c:\cockpit\nick\cockpit\cockpit\handlers\lightPower.py", line 91, in _updater
    light.lastPower = queries[light].result()
  File "c:\python37\lib\concurrent\futures\_base.py", line 425, in result
    return self.__get_result()
  File "c:\python37\lib\concurrent\futures\_base.py", line 384, in __get_result
    raise self._exception
  File "c:\python37\lib\concurrent\futures\thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "c:\cockpit\nick\cockpit\cockpit\devices\microscopeDevice.py", line 345, in _getPower
    return self._proxy.power
  File "c:\python37\lib\site-packages\pyro4-4.75-py3.7.egg\Pyro4\core.py", line 277, in __getattr__
    return self._pyroInvoke("__getattr__", (name,), None)
  File "c:\python37\lib\site-packages\pyro4-4.75-py3.7.egg\Pyro4\core.py", line 476, in _pyroInvoke
    raise data  # if you see this in your traceback, you should probably inspect the remote traceback as well
ValueError: could not convert string to float: b'?p'

Which is I think a consequence of the power not being set at startup even if the laser in enabled.

juliomateoslangerak, 2021-06-02:
I have the same error

carandraug, 2021-06-03:
For those who don't know DeepSIM, its 561 line is a Coherent Sapphire laser.

carandraug, 2021-06-03:
> Which is I think a consequence of the power not being set at startup even if the laser in enabled.

I don't think that's what's happening. I just tried it directly and works:

```
>> s = SapphireLaser('COM2')
>> s.initialize()
>> s.power
0
```

juliomateoslangerak, 2021-06-03:
In my case it is with the 488 deepstar. I will check tomorrow if it is exactly the same error.

carandraug, 2021-06-03:
> In my case it is with the 488 deepstar. I will check tomorrow if it is exactly the same error.

Must be a different one. The bottom of the error message refers to the command `b'?p` which the sapphire command to query the power. The deepstar laser has no such command.