python-microscope/microscope #210: Client import of properties from the proxy are integers instead of properties

gjover, 2021-06-04:
When a client is created, connecting to a microscope device, the properties passed from the  proxy are integer attributes instead of properties.
This makes that any change on these attributes do not affect the proxy property nor the other way around.

carandraug, 2021-06-08:
You are right and yours seems the right fix.  However, why are you using this?

Does anyone know why we even have this module?

`microscope.clients.Client` can be replaced with `Pyro4.Proxy`

`microscope.clients.DataClient` I'm not sure what's the use case here.

gjover, 2021-06-08:
It is convenient to test the DS from python console.
I was wondering if it could be used for abstracting the communication in other packages as the microscopeDevice in cockpit.
But may be this class is not used anyhow.