python-microscope/microscope #211: Issue with multiple Andor iXons on one computer.

iandobbie, 2021-06-10:
Had a very strange issue with cryosim software today. We were attempting to update microscope  to current master branch. Everything appeared to  work except ONE of the andor cameras. Both cameras are connected to a single system. The old software worked fine, but the current master failed consistently on one camera. Selecting only a single camera worked with one but failed with the other.

I think it might be the floating device code and one camera is picked up first and works whereas the other is picked second and fails. Unfortunately I didn't think to turn off or disconnect each camera in turn to see if this is correct. I am going back on Tuesday and will see if I can find the root of the issue to help solve it.

carandraug, 2021-06-11:
When you say fails to start do you mean it fails to initialize the device server for the second camera? Or does it fail later during enable or something like that? Do the logs say where it fails?

iandobbie, 2021-06-11:
I didnt have time to properly explore this as I had to leave and needed to leave a working system. I am back next week and will investigate in more detail. 

I realise that this is not a fundamental microscope issue as the recently updated deepsim setup is basically identical, 2 ixons connected to a single computer. The fact that the hardware works fine with the older microscope with the same config makes me think it is not HW related. I guess it might be some weird interaction of microscope versions and configs?

iandobbie, 2021-06-11:
I realise I didn't answer you question. The second camera did not initialise and so the deviceserver was restarting the process every 5 seconds. I didn't capture the error output but will make a serious effort on this issue on Tuesday.

iandobbie, 2021-06-16:
fixed with pull #212

closed 2021-06-16 by iandobbie