python-microscope/microscope #224: Andor: avoid to send a software trigger during an acquisition

gjover, 2021-08-18:
If a software trigger is sent during an acquisition, an DRV_ERROR_ACK error is risen.
To avoid this, an sleep time is added for the remaining time until the end of the acquisition cycle.

carandraug, 2023-01-10:
What to do when `Camera.trigger` is called but the camera is still not ready to process a new trigger is a very good question and something we haven't defined yet.

The proposal here is to wait until the camera is ready and then send the trigger. I'm not sure about that behaviour. That's certainly not the behaviour with hardware triggers, which maybe would be nice to mimick. Would be nice to have a discussion about the different behaviours and pros and cons when applied to all cameras.

juliomateoslangerak, 2023-01-10:
Considering the time it takes for a camera to be read, I think that, for software triggers, it is a reasonable behavior to wait until the camera is ready.
For hardware triggers, I had this problem before (but without error raised) and the only way around was to do the proper upfront calculation of the timings (hence one of the reasons why it is useful to know if the camera has a global or rolling shutter).
Alternatively, the executor should implement responding to input triggers:
- pause until trigger X
- wrap table on trigger X
- move to table position N on trigger X
- ...