python-microscope/microscope #229: Set exposure time on TriggerTargetMixin

gjover, 2021-09-17:
How would one set the exposure time on a TriggerTargetMixin device?
Would it be worth to add setting exposure time on the TriggerTargetMixin abc class as it is in the Camera Class?

carandraug, 2021-09-18:
I think there's some misunderstanding. The `TriggerTargetMixin` is [mixin class](https://en.wikipedia.org/wiki/Mixin) which is meant to be mixed in device classes. It is mixed in the classes of devices that can receive triggers but configuring the action is specific to the device type and so is done by the device class only.

In the specific case of cameras, the `Camera` class already has the `TriggerTargetMixin` class mixed in (because all cameras can be targets of tiggers even if only software triggers). So use the `Camera.set_exposure_time` method, no need to also specify it on the `TriggerTargetMixin` class.

If that makes sense, please close this issue.

gjover, 2021-09-20:
Yes, I understand that it is a mixin class.
I was thinking that in, terms of cameras and lights, when one triggers a device it is always for a certain amount of time.
In that sense I was proposing to generalize the implementation of set_exposure_time, moving the implementation from the camera to the TriggerTargetMixin to share the code with any other device class using it.
But it is true that for some classes the triggered action may have more configuration parameters than just the exposure time.

carandraug, 2021-09-20:
Having a method to configure the triggered action in the mixin class could make things cleaner, I guess it depends how that is done. It couldn't be just `set_exposure_time` since there are other devices types that are trigger targets where it would make no sense such as deformable mirrors. My guess would be that differences between each trigger action is so large, that they might as well be done on the individual device types. But if you can come up with something general that would be great.