python-microscope/microscope #236: Thorlabs filter wheel fails to update

thomasmfish, 2022-03-10:
This was fixed at B24 with Ian's help a while back (see PR #215). That said, it may be something that can be done a little more elegantly?

carandraug, 2022-03-11:
Regarding the fix in #215, if the root issue is that getting the position fails because the device is still moving, then simply trying a second time could still fail if the move is long.

Is there a command in the filterwheel that returns when it has finished moving?

thomasmfish, 2022-03-14:
I haven't yet had a chance to delve into this more but I'm hoping so. Does microscope have a way to expose the output from devices to the command line?

thomasmfish, 2022-03-14:
Okay, I've fixed the issue in a much more comprehensive way now. The issue was empty lines being returned, presumably due to a change in polling frequency or something, which ended the while loop before it had even echoed the command. I've improved the two methods involved to:

1. Never return fully empty strings as readings. There should be at least some non-whitespace characters there before it is considered a response.
2. Catch ">" rather than stripping it and waiting for "". I'll link the PR now so you can take a closer look.

thomasmfish, 2022-03-14:
Just linked #215 with the latest changes

PR #215 (2022-06-08): fix thorlabs filter wheel position read error
Fixes #236
closed 2022-06-08 by iandobbie