python-microscope/microscope #239: Unsupported temperature sensor in xiB-64

jacopoabramo, 2022-04-13:
Greetings,

I just experimented with the free-running acquisition using python-microscope to control my Ximea xiB-64 camera (the snippet code is presented in image.sc [here](https://forum.image.sc/t/free-running-acquisition-example-with-python-microscope-ximea-device/65740). This works, but I had to apply a small fix in my local microscope installation because whenever I created a new camera object the initialization failed due to the fact that my camera returned error 26 ("Not implemented") when adding the setting related to the temperature sensor. I wanted to mention this as the fix I did was pretty straight forward:

```python
try:
    get_temp_method()
except xiapi.Xi_error as err:
    if err.status != _XI_NOT_SUPPORTED and err.status != _XI_NOT_IMPLEMENTED:
        raise
```

With `_XI_NOT_IMPLEMENTED = 26`.

carandraug, 2022-04-18:
What version of xiAPI are you using?

Some time ago I reported to them that when missing, some sensors raised `XI_NOT_SUPPORTED` while others returned `XI_UNKNOWN_PARAM`. Then on version 4.21.19, to fix this they made them all raise `XI_UNKNOWN_PARAM` and commented that their choice was not very intuitive and that they might review it later. I guess they did it now and decided on `XI_NOT_IMPLEMENTED`. See original issue on python-microscope/vendor-issues#6

carandraug, 2022-04-18:
Thank you for the fix. I have pushed a slightly modified version of your patch. Closing as fixed.

closed 2022-04-18 by carandraug

jacopoabramo, 2022-04-18:
Hi @carandraug , I checked the Python wrapper version and it's 4.22.00 (which I assume it's the same version of the installed xiAPI as I installed both together at the time).