python-microscope/microscope #241: Device server on Raspberry Pi

rreeves60, 2022-05-02:
I just wondered if a device server could run on Raspberry pi. If this has not been tested I will attempt to compile the python script on rasperian

rreeves60, 2022-05-02:
I work for Martin booth

rreeves60, 2022-05-02:
I also see some work has been done for PiCam will also check this out.

closed 2022-05-02 by rreeves60

iandobbie, 2022-05-02:
The device server will definitely run on a raspberry pi and some hardware is supported typically hardware that talks serial or serial over USB. Other USB devices might be supported but they may lack drivers for linux in general or the Pi in particular. One other thing to consider is that devices with significant IO or processing requirements maybe slow and even the new fast Pi4 is not very fast compared to the multi core desktop or laptop machine.