python-microscope/microscope #243: Adding `TriggerType.OFF`

jacopoabramo, 2022-05-24:
Greetings,

I already mentioned in an image.sc [thread](https://forum.image.sc/t/free-running-acquisition-example-with-python-microscope-ximea-device/65740/4) that I wanted to use microscope to control an industrial-grade Ximea camera with a continuous frame capture. After some thought, the only way to actually manage to do it was to add a new value to the `TriggerType` enumerator.

I wanted to know if this is a feasible change to add in the main fork or if its outside the scope of how microscope should be used. You can check the commit [here](https://github.com/Eggeling-Lab-Microscope-Software/microscope/commit/e1ee225d17b084c01d60aa4e8660017a090173d7). Not a big change but it served my objective quite nice.

carandraug, 2022-05-25:
I already replied on the image.sc thread but I'll add some more detail here. There's also some assumptions from me on what it is that you're trying to achieve, so correct me if I'm wrong.

1. The goal here is to have `enable` start acquiring images of a given exposure time and have `disable` stop.
2. This behaviour should be mapped to `(TriggerType.SOFTWARE, TriggerMode.STROBE)` which is currently not implemented in `XimeaCamera`. 
3. On the Ximea side things, the wanted behaviour is `(XI_TRG_SOURCE.XI_TRG_OFF, XI_TRG_SELECTOR.XI_TRG_SEL_FRAME_START)`.
4. The way `XimeaCamera` handles trigger configuration is by mapping Ximea's `trigger_source` to Microscope's `TriggerType`, and Ximea's `trigger_selector` to Microscope's `TriggerMode`.
5. We already map Microscope's `TriggerType.SOFTWARE` to `XI_TRG_SOURCE.XI_TRG_SOFTWARE` so we can't also map it `XI_TRG_SOURCE.XI_TRG_OFF`
6. The solution being proposed is to add a new `TriggerType.OFF` so that things just work automatically without much changes to the `XimeaCamera` internals.

I really don't think this is the right approach. The issue is in our simple mapping of Microcope's trigger type/mode to Ximea's trigger source/selectors are just not fit for purpose anymore. So the solution is to change those internals and not to extend the public interface.

carandraug, 2022-05-25:
Putting it a on slightly different manner. What would `TriggerType.OFF` add to the public interface that is not already possible with the existing interface?

iandobbie, 2022-05-25:
Overall this is showing there is a fundamental issue of incompatible and somewhat arbitrary triggering control mechanisms in different camera APIs.

What I believe is being attempted is to run the camera in a free run mode, which microscope has never really supported. If we are going to add a trigger type I think it should be something along the lines of TriggerType.FREE_RUN as many cameras APIs implement a free run mode but in different manners. Then it is up to the microscope device to set the relevant device specific parameters to achieve this aim. 

This still adds to the existing API but in a much clearer manner and will not generate a weird special case where you have to set trigger to OFF on Ximea cameras but some other gymnastics on other  cameras.

iandobbie, 2022-05-25:
The TriggerMode.STROBE is from the Andor cameras that keeps exposing a single frame while the hardware trigger is high, eg provides external control of the individual frame length, this is a different mode from FREE_RUN where the camera determines the individual frame length internally (presumably from the exposure time setting) but it keps taking new images until told to stop.

carandraug, 2022-05-25:
> The TriggerMode.STROBE is from the Andor cameras [...]

This does not look right. `TriggerMode.STROBE` is not handled by `AndorAtmcd` or `AndorSDK3`. Maybe one of Andor SDK's does have a mode named strobe but it's not handled in Microscope.

PVCam also has a mode named "strobe" which is when an external trigger starts each exposure in a sequence. This is very different from your description of Andor's "strobe" mode. Also, PVCam "strobe" mode seems to map to AndorSDK2 "external" mode. My point with this is that we can't try to use the vendor name in the Python Microscope API or we'll just end up with clashes of modes. We need to use our own trigger modes/types and then on each camera class map it to whatever each vendor calls it.

I have made a table above with what I had in mind for each trigger type/mode. Without trying to reuse names from the many many different and incompatible vendors, can you fix the table to add the new configurations that are not supported?

jacopoabramo, 2022-05-26:
Hi both, thank you for the replies!

I'll just address the point raised from @carandraug:

> Putting it a on slightly different manner. What would `TriggerType.OFF` add to the public interface that is not already possible with the existing interface?

First of all, I agree that my proposal is a bit far fetched and only limited to the current use case scenario I've encountered so far (having a camera in free run condition), and even while opening this issue I was aware that even if it was a single line of code it would have made a bigger change than it looks like. But by only relying on the microscope documentation (and, of course, I might have missed something too in reading it) and source code, and in order to avoid having to drastically change the Ximea interface for my needs, I thought this was the most "easy" approach.

I definetely think that the table you proposed though is a great indicator on how to correctly use the `TriggerType` and `TriggerMode`, and adding it to the documentation would be of great help. I just have a question regarding the combination `SOFTWARE` - `BULB`. From my understanding after reading both yours and @iandobbie comments, I'm guessing that:

- the camera starts an exposure cycle
- a call to `trigger` fetches an image from the camera

I'm not familiar with the Andor cameras so maybe you could clarify this for me (just in case in the future I may need to use this approach).

Still, if I may be somewhat bold, I find the usage of the `STROBE` keyword a little confusing, especially when in relation to the table presented above. Maybe taking into account Ian's suggestion, you could consider using `FREE_RUN` as a substitute for `STROBE` in the supported `TriggerMode` rather than adding it as a `TriggerType`. From the table it seems that it fits better in the description of how the camera should behave in case of a trigger event. Of course it's just food for thought.

iandobbie, 2022-05-27:
Sorry I am obviously totally miss remember the facts of this. I will make sure I know what I'm talking about next time. I agree about the names, every manufacture calls them something different.  We just need to define the behavior for each state and then the relevant device needs to map that onto the specific settings for that camera or range of cameras.

carandraug, 2022-05-30:
> I just have a question regarding the combination `SOFTWARE` - `BULB`. From my understanding after reading both yours and @iandobbie comments, I'm guessing that:
>
> * the camera starts an exposure cycle
> * a call to `trigger` fetches an image from the camera

Kind of. In the combination `SOFTWARE` - `BULB`, when you call `enable` it starts exposing and when you call `disable` it stops exposing. Afterwards, the image will be available in the client/queue, but there's no need to call `trigger`.

The `trigger` method sends *one* trigger and so only makes sense for actions that happen in response to *one* trigger: `START` (start a predefined sequence) and `ONCE` (acquire one image). But sending *one* trigger makes no sense for actions that happen while the device is being triggered: `BULB` (expose while being triggered) and `STROBE` (acquire while being triggered) because there's no method to start/stop a trigger. Maybe we could introduce such methods? I thought that we could get away with using `enable` / `disable` but maybe that's confusing and slow (some cameras take a while to enable/disable). Opinions?

> [...] I find the usage of the STROBE keyword a little confusing, [...] consider using FREE_RUN as a substitute for STROBE in the supported `TriggerMode` rather than adding it as a `TriggerType`.

Sure. We are using an enum and we can use alias (we already do this, `TriggerType.HIGH` is an alias for `TriggerType.RISING_EDGE`). See:

```
>>> import enum
>>> class TriggerMode(enum.Enum):
...     ONCE = 1
...     BULB = 2
...     STROBE = 3
...     FREE = 3
... 
>>> TriggerMode.STROBE == TriggerMode.FREE  # equality works
True
>>> tmap = {TriggerMode.STROBE: "some method"}
>>> tmap[TriggerMode.FREE]  # also works as keys
'some method'
```

The reason why this mode is named `STROBE` is that it is the mode that intermittently triggers an action in the device. In the context of a camera, that action is exposing the sensor to acquire one image. So strobe means that intermittently acquiring images. These trigger modes and types names are not only for cameras. They are shared between all devices, so deformable mirrors and light sources also use them.

jacopoabramo, 2022-06-07:
Hi both,

apologies for the late reply - I was on vacation until yesterday. I think that @carandraug has clarified what I wanted to implement with using the current interface. I'll follow your suggestions and make the changes on the Ximea interface rather than the common one. I'll close the issue.

jacopoabramo, 2022-06-20:
Hi everyone,

as requested by @carandraug , I'm reopening this issue just to notify the changes I did for the Ximea interface in relation to the trigger management. You can check the changes in [this commit](https://github.com/Eggeling-Lab-Microscope-Software/microscope/commit/b21ee43b93c52c280549510929503e2b97e8ae76) on my microscope for.