python-microscope/microscope #249: Need to make mosaic

Divyadeep00, 2022-07-07:
Hello everyone, 

I have leica microscope with camera. Model no. Dm 6000 B.  I want to make a image,  which represent full slide (mosaic of images of every unit area). But I want this to be automatic.  Is there any script or how to make such script.

Looking forward to hearing from you soon.

carandraug, 2022-07-07:
Such a script can be written but this is typically done through some GUI program (such as [cockpit](https://www.micron.ox.ac.uk/software/cockpit/)). However, there is currently no support for that camera.

iandobbie, 2022-07-07:
We have not investigate the Leica frames at all but I'm afraid our previous efforts with major manufactures frames have been a total dead end. Both Zeiss and Olympus require an MTA to be signed to get access to the relevant libraries to interface with their scopes. We are not willing to do this as it restricts us distributing source to connect to their  driver. 

If this is different for Leica, thats great news, I would love to hear that.

Divyadeep00, 2022-07-07:
But let us make this for Leica and Olympus because these two are widely used microscope frames, at least in India; if these are commonly used in the entire world, then this python-microscope will flourish very quickly and be helpful for many educational institutes because they don't have to pay for such costlier software of Leica. 

Thank you

Divyadeep00, 2022-07-07:
yeah, I can control the microscope from Leica LAS V4.2 software, but the mosaic system is not there.

carandraug, 2022-07-07:
> I do not have much programming knowledge, so maybe I am not the right person for this. By the way, thank you for your reply.

What @iandobbie means with "require an MTA to be signed to get access to the relevant libraries to interface with their scopes" is that the only way to control such microscope is if Leica tells us how and they only tell us if we don't write software for others which makes the whole thing pointless.

Anyway, I thought this discussion really belongs to issue #248 and that this was about having, probably on documentation, a script example for whole slide acquisition. If not, then this is a duplicate of #248.

Divyadeep00, 2022-07-07:
Yeah, I raised this issue only for the script for the whole slide acquisition. and issue #248 was to know the support of Python microscope on Leica.

iandobbie, 2022-07-07:
You might find some way to do this on image.sc.  A thread to start with might be 
https://forum.image.sc/t/micro-manager-for-leica-dm6000b-with-ctr6000-controller/68582