python-microscope/microscope #252: Support for thorlabs ELL6, 9 and 12

juliomateoslangerak, 2022-09-26:
I'm trying to add support for Thorlabs ELL Multi-Position Sliders. Models ELL6, 9 and 12.
These devices are controlled through a serial interface. 

- These linear stages seem to respond to the command "ma" (move absolute) to go to a specified position. However, I cannot find in the documentation how to specify the position. According to the documents you have to provide position X pulses_per_unit, but pulses per unit for these devices is 0. I figured out that you have to directly specify mm.
- You need to figure out how many mms to move from the pitch size, that is the distance between the filters. Yo may get this info calling some info command. And while my device is reporting a pitch size of 31mm the slider will not work like this. You have to command it to move 32mm in stead, but it will reach the position at 31mm. I figured this out by looking at Thorlabs own software. I contacted thorlabs about this but the answers are not clear to me. 
This is an example of thorlabs Elliptic communication with the device when I connect, home and cycle through the positions by clicking in the buttons to move to positions 1, through 4.

It is just disturbing to not know where that extra mm comes from.
```
Connecting...
Scanning for devices
Tx: 0in
Rx: 0IN091090010620200801005D00000000
Tx: 1in
1 device found
Get Jogstep Size...
Tx: 0gj
Rx: 0GJ0000001F
Requesting Motor1 info...
Tx: 0i1
Rx: 0I1100000FFFFFFFF00B2008D
Get Device Status:
Motor ID 1
Loop State On
Motor State Off
Current 0.00A
Fwd Freqency 82.8kHz
Rev Frequency 104.5kHz
RampUp 65535
RampDown 65535
Requesting Motor2 info...
Tx: 0i2
Rx: 0I2100000FFFFFFFF00B2008C
Get Device Status:
Motor ID 2
Loop State On
Motor State Off
Current 0.00A
Fwd Freqency 82.8kHz
Rev Frequency 105.3kHz
RampUp 65535
RampDown 65535
Homing device ...
Tx: 0ho0
Rx: 0PO00000000
Move device to 0.0 mm...
Tx: 0ma00000000
Rx: 0PO00000000
Move device to 32.0 mm...
Tx: 0ma00000020
Rx: 0PO0000001F
Move device to 64.0 mm...
Tx: 0ma00000040
Rx: 0PO0000003E
Move device to 96.0 mm...
Tx: 0ma00000060
Rx: 0PO0000005D
```

iandobbie, 2022-09-26:
I understand your frustration but in our experience hardware control often has extreme weirdness and manufactures are very reluctant to fix the firmware or interface code.

I suggest we document it carefully, including something in the code so any future person working on it (maybe even your future self) can see why you are doing something "wrong", and then not worry about it too much. Having some kind of issue with the manufacturing is good practice as well. If they give you some kind of ticket number or similar you could add that to the documentation for the issue.

juliomateoslangerak, 2022-09-30:
Thorlabs support is frustrating indeed.
Maybe we can have some luck looking in here https://micro-manager.org/ThorlabsElliptecSlider