python-microscope/microscope #257: Olympus DP23 camera support

Mygyth, 2022-11-23:
Hi, sorry if this is the wrong spot for this, im very new to programming.

Im trying to access live feed from an Olympus DP23 so that i can build a machine learning algorithm to automatically ID certain features. How possible is it that you can add this camera to the functionality of this project?

iandobbie, 2022-11-28:
Expanding hardware support is one of the main goals for Python-Microscope, however there are two major hurdles for many add-ons produced by all of the major microscope companies. 

1) They often don't provide an open API and freely distributable control libraries (dll's on windows)

2) Hardware invariably doesn't actually do exactly what it says in the documentation, so having a real example to test on  is essential.  We don't have the budget to buy and test every piece of hardware. In general the hardware control we have is either hardware contributors own or in a few cases items that companies have leant to allow drivers to be written.

To date, our experience with the major microscope manufactures is that releasing full specs and documentation and then allowing free distribution of the code is the biggest hurdle.  For some reason they see to think that it is "commercially sensitive" to know exactly what modes their camera has, and how to switch between them. They appear to be deaf to the idea that providing proper open source access might increase their sales.

BiologyTools, 2022-11-28:
I once automated Olympus FV1000 using GUI Automation to take images & move the focus. The prior stage came with a .NET library so it was the easy part. I suppose you could do something similar. Grab the live feed image with GUI Automation element properties & then automate clicking the take image button. For my project I used C# but there's probably good python libraries to do the same.

iandobbie, 2022-11-29:
This approach has a number of disadvantages. The code is then completely reliant on a piece of external code. The system has to run a large chunk of probably slow unused GUI code. This is also extremely fragile as  minor changes in the GUI or menu structure are going to completely break the control.

Fundamentally, I don't feel this is a sensible approach to hardware control and I would not utilise this mechanism, just buy a different camera. 

I would not implement such an approach, but if other peopel