python-microscope/microscope #258: pvcam bulb mode does not work

carandraug, 2022-12-17:
Me and @VroniPfann were having issues setting up Photometrics Kinetix for bulb mode. We would `set_trigger(TriggerType.HIGH, TriggerMode.BULB)` but it would still behave as `TriggerMode.ONCE`. It appears that we only support the legacy PVCam trigger modes which should not be used in the sCMOS cameras

In PVCam, exposure modes are a bunch of enums. Previously, it could be *one of*:

* `TIMED_MODE`
* `TRIGGER_FIRST_MODE`
* `STROBED_MODE`
* `BULB_MODE`
* `VARIABLE_TIMED_MODE`

These are now considered legacy. Since PVCam 3.0.1 there are extended exposure modes. Instead of *one of* those exposure mode, PVCam now has other exposure modes that need to `OR`ed with the new exposure out  modes. The legacy modes still appear to work but just not the way we expect. I'm not sure why we see the behaviour we see but the following change fixe the issue for us:

```
--- a/microscope/cameras/pvcam.py
+++ b/microscope/cameras/pvcam.py
@@ -261,6 +270,7 @@ FLASH_MODE = 4
 VARIABLE_TIMED_MODE = 5
 INT_STROBE_MODE = 6
 MAX_EXPOSE_MODE = 7
+EXT_TRIG_LEVEL = (7 + 3) << 8
 Extended = 8
 camera = 9
 The = 10
@@ -1167,7 +1177,7 @@ TRIGGER_MODES = {
     TRIG_VARIABLE: TriggerMode("variable timed", VARIABLE_TIMED_MODE),
     TRIG_FIRST: TriggerMode("trig. first", TRIGGER_FIRST_MODE),
     TRIG_STROBED: TriggerMode("strobed", STROBED_MODE),
-    TRIG_BULB: TriggerMode("bulb", BULB_MODE),
+    TRIG_BULB: TriggerMode("ext trig level", EXT_TRIG_LEVEL),
 }
 
 PV_MODE_TO_TRIGGER = {
 ```

We never check what exposure modes are actually supported by the camera. It seems we should be checking that dynamically (PVCam lets us do that) and adjust what modes we use based on that.

iandobbie, 2022-12-19:
Sounds like a sensible solution. I guess the only worry is that updates break working systems. The ones that come to mind are the Oxford systems based around older photometrics cameras like the Evolve Delta's, or even the exi Blue and green (are those the right names?)  that we occasionally used in Oxford. There is also the Prime BSI that was on the AO system in engineering.

carandraug, 2022-12-19:
@juliomateoslangerak you mentioned you use bulb mode on a photometrics kinetix. What pvcam version are you using? And are you changing it for bulb mode?

VroniPfann, 2023-04-24:
@carandraug I think we fixed this issue just before Christmas. The Kinetix is working fine in bulb mode now. The code can be found on my vortran_laser branch here: https://github.com/VroniPfann/microscope/tree/vortran_laser/microscope