python-microscope/microscope #259: PIPython, a library to control Physik Instrumente devices

juliomateoslangerak, 2023-01-05:
I just came across this https://github.com/PI-PhysikInstrumente/PIPython
It would be interesting to use it for the PI stages

carandraug, 2023-01-06:
My experience so far with Python wrappers from vendors is that they're not worth it. They are undocumented so every you have an issue you need to check the original documentation so you end up having to be aware of the oddities of their platform but also any extra oddity they have introduced in the Python wrapper.

Also, PIPython is non-libre software (but neither are the vast majority of the shared libraries we use).

carandraug, 2023-01-06:
This is effectively about adding support to PI stages as discussed on issue #196 . Closing as duplicate.

closed 2023-01-06 by carandraug