python-microscope/microscope #261: incorrect use of WinDLL

carandraug, 2023-01-09:
We use `ctypes.WinDLL` to load all shared libraries in Windows but that's wrong. We should be using `ctypes.CDLL` even on Windows for some libraries.

We have been using `WinDLL` under the incorrect assumption that's what one needs to do in Windows. However, the choice is not about the underlying OS, it's about the calling convention used by the functions. We should be using `WinDLL` to access functions that use the `stdcall` calling convention and `CDLL` to access functions that use the standard C calling convention (called `cdecl` on Windows).

If I understood this correctly, we need to check the header files and see if in Windows the functions are defined with `__stdcall` (may happen indirectly such as when the `WINAPI` macro is used). If not, then we should be using `CDLL` even on windows. It seems that using the wrong one will not immediately lead to issues but may lead to weird unpredictably issues later.

I've checked the header files and seems we need to change for Alpao and BMC mirrors, for Hamamatsu cameras, and for Linkam stages:

| device (module) | checked header | needs fix |
| --- | --- | --- |
| alpao mirrors (`microscope/_wrappers/asdk.py`) | yes | probably |
| BMC mirrors (`microscope/_wrappers/BMC.py`) | yes | probably |
| hamamatsu cameras (`microscope/_wrappers/dcamapi4.py`) | yes | probably |
| mirao mirrors (`microscope/_wrappers/mirao52e.py`) | yes | no (`MIRAOCALL` expands to ` __stdcall`) |
| pvcam cameras (`microscope/cameras/pvcam.py`) | yes | no (`PV_DECL` expands to `__stdcall` on Windows) |
| andor SOLIS (`microscope/cameras/atmcd.py`) | yes | maybe (uses `WINAPI` on one version but nothing on other) |
|andor SDK3 cameras (`microscope/cameras/_SDK3.py`) | yes | no (`AT_EXP_CONV` expands to `WINAPI`) |
| linkam stage (`microscope/stages/linkam.py`) | yes | probably |

However, I no longer have access to any of these. Could someone who does have access test it?

dstoychev, 2023-01-11:
Tested the ALPAO mirror wrapper. Replacing `WinDLL` with `CDLL` on line 29 works fine!

https://github.com/python-microscope/microscope/blob/55138bf2c9aeab2be88b04872c6dc097ab629fbb/microscope/_wrappers/asdk.py#L29

carandraug, 2023-01-12:
Thank you for testing Danny. I've pushed e837dfc8bfa3. I've also asked @thomasmfish by email whether he could check the Linkam stage.

iandobbie, 2023-04-11:
Tom Fish says that Diamond is using CDLL for the linkam stage there and it works for them.