python-microscope/microscope #263: Massive Debug log when run in simulated mode with image.

iandobbie, 2023-01-12:
I am repackaging cockpit_sim from the cockpit-distibution with the latest cockpit and microscope on a mac. Once packaged the application works if run directly from the command line (> ./dist/cockpit_sim.app/Contents/MacOS/cockpit_sim) but generates huge amounts of debug messages like 

2023-01-12 15:30:45,070:simulated_setup_from_image (microscope.abc):DEBUG:PID 46450: Fetching data from device.

These come roughly every ms! When run from the finder or with open dist/cockpit_sim.app it crashes I assume because of the spammed text output.

iandobbie, 2023-01-12:
Oh this only happens once the camera is enabled.

Maybe I should just turn off debug messages?

iandobbie, 2023-01-12:
This is not the cause of the startup issue as this only occurs after the camera is enabled. But it does need to be fixed.

carandraug, 2023-01-13:
> Oh this only happens once the camera is enabled.

Those are debug message from `DataDevice._fetch_loop`. They were added recently in f3ee7f6ed57577dcc86983ce9fcc81f78a3e4de4 . It logs that it attempted to fetch an image from the camera.

In the fetch loop, it tries to fetch an image from the camera. If there's nothing, it returns `None` and tries again after a few milliseconds. Without a log message when it tries and fails to get an message, you're a bit blind debugging it. But maybe we could cut on the messages a bit. Or comment it out so people debugging that part can enable them. Or maybe we could increase the sleeping time from 1 millisecond to 100 milliseconds. 

> Maybe I should just turn off debug messages?

If you're not debugging, definitely.

iandobbie, 2023-01-13:
Found the real root cause, which is that the logging level in cockpit-distribution was set arbitrarily with logging_level = 1. I changed this to import the defs from logging and set to logging.WARNING which I think is probably the best solution.

closed 2023-01-13 by iandobbie