python-microscope/microscope #266: Configuring startup settings on a device in the server

juliomateoslangerak, 2023-01-13:
I need to be able to configure the initial settings of a device. Users do not want to go through a list of settings to chose the different options that are always the same at startup. In my case this involves 5 settings per camera, including a 2-3 applies. 

According to the documentation this is achieved through adding a function to the DEVICES list in the config file. However this is not working under Windows (at least for the moment). #179 

Independently of the need of sending a function to the device server, I was wondering if it would be possible to add another dict argument to the device to configure the settings after initialization. For example ```initial_settings = {"read_speed": "100MHz", "sensor_temperature": "-10",...}```.
At the moment, the ```conf``` argument is easily confused with this purpose.

carandraug, 2023-01-13:
I guess something like that could be added, but since you are using cockpit, why not specifying it on the depot file? Something like this should work (the settings named need to be indented):

```
[south camera]
type: cockpit.devices.microscopeCamera.MicroscopeCamera
uri: whatever url
settings:
    some-option-name = some value
    some-other-option = some other value 
```

juliomateoslangerak, 2023-01-16:
That's great, I was not aware of this feature. I looked for it and I cannot find it documented. Is it?
Just note that the ```=``` is not working. You have to use the colon ```:``` notation to separate key and value. This is my example depot for a camera:
```
[Green-Cam]
type: cockpit.devices.microscopeCamera.MicroscopeCamera
uri: PYRO:AndorSDK3@10.6.19.31:8001
transform: (1, 1, 1)
triggerSource: cRIO
triggerLine: 4
settings:
    aoi_height: 1024
    aoi_width: 1024
    aoi_left: 513
    aoi_top: 513
    pixel_readout_rate: 100 MHz
    simple_pre_amp_gain_control: 16bit (low noise & high well capacity)
    trigger_mode: External Exposure
```
I guess the question of weather we need this kind of functionality on microscope too remains

carandraug, 2023-01-16:
> That's great, I was not aware of this feature. I looked for it and I cannot find it documented. Is it?

You are right, it is not documented. But that's just being consistent with the rest of Cockpit features ;)

> Just note that the `=` is not working. You have to use the colon `:` notation to separate key and value. 

The problem is the space between the `=` and the setting name (the regex that does this uses a greedy modifier). I've opened MicronOxford/cockpit#840 about and will fix it soon.

> I guess the question of weather we need this kind of functionality on microscope too remains.

I would rather we fixed #179

iandobbie, 2023-03-03:
I will document the "setting:" feature in the cockpit depot file.

iandobbie, 2023-03-03:
It was kind of docuemented but not properly. It didn't mention the need for the settings: keyword at the start, which kind of defeated the purpose. The text is in  6e87692ee46b0c04348ca6c7f1f5a4ff9d30b5af

thomasmfish, 2023-03-03:
I just tried this on our microscope and I got messages saying the settings weren't recognised. I was trying setting the settings in the documented examples but I may have been doing something wrong, or it could be dependent on a more recent version of Microscope. When was this feature added, and how can I check what settings are settable? It seems that the configurable settings don't align 1:1 with what is displayed in the settings panel and there seems to be a bit of a range of formatting styles, so I'm a bit lost.

iandobbie, 2023-03-03:
What you you trying to set? The examples here are for an andor cMOS camera,  your EMCCDs will be a bit different.

thomasmfish, 2023-03-06:
Ah, I didn't realise that. The readout mode, fan state, and temperature are the thing we want to set. Is there a way to figure out from the code what is available and how it should be written?

thomasmfish, 2023-03-06:
I have it working now - I thought there was some messing with underscores that I'd need to do, and I hadn't been able to get access to test, but I was able to just copy them directly from the settings panel. Thanks!

iandobbie, 2023-03-08:
Great, closing as fixed, and I have added some documentation as well

closed 2023-03-08 by iandobbie