python-microscope/microscope #268: Redirect logging messages from local logger to external one

jacopoabramo, 2023-01-23:
Greetings,

I'm currently implementing python-microscope within ImSwitch, a software which has its own internal logger which I wanted to take advantage of to keep consistency with the logging messages. Would it be possible to redirect the microscope's logging messages to a different one? And if so, how? It's not really an "issue" per se, but I could not find the discussion tab in the repo.

Thank you!

carandraug, 2023-01-23:
Microscope simply uses Python's logging module. You "just" need to set the handlers to wherever you want the logs to go. The how depends on where you're starting the device server and what logging system does ImSwitch is using.

jacopoabramo, 2023-01-23:
Great, then I'll dive deep into how to do this in ImSwitch. Thanks! Closing.

closed 2023-01-23 by jacopoabramo

iandobbie, 2023-01-23:
As far as I understand it you just need to replace the log_handler in device_server.py somewhere around line 350. I have been playing with this to add a defined path rather than the cwd see #267