python-microscope/microscope #269: Some stage sizes are not known until stage runs init

iandobbie, 2023-01-25:
Cockpit struggles as it doesn't know the stage size of many stages, including the Ludl until the stage has been init'd and moved to its limits to find it own size.

carandraug, 2023-01-26:
I don't understand what the issue is. Can you expand what you mean with "struggles"? Cockpit calls enable right at the start (which triggers all this) and even checks if enable will cause that move to display a message at start.

iandobbie, 2023-01-26:
I just wanted to check that the current behavior was fully working before closing this issue. I believe that it was all fixed with the additional function in microscope and changes in cockpit but I need to actually test this functionality. 

This was to separate this issue from the fact that the Ludl controller returned useless info if the controller was off or otherwise no connected.  I will check probably on Friday and close with a relevant comment if it is truly fixed.

iandobbie, 2023-01-26:
There remains a slight problem, on startup the stage homes but it doesn't appear to update the limits properly so the cockoit limits are mad. Not sure if this is a microscope or a cockpit problem, or more likely something in the interface.

iandobbie, 2023-01-26:
This is definitely a cockpit problem and I am now fixing it, marking this issue as closed. Cockpit changes are now in https://github.com/MicronOxford/cockpit/issues/842

closed 2023-01-26 by iandobbie