python-microscope/microscope #271: Clarification on the "transform" parameter

jacopoabramo, 2023-02-14:
Greetings,

To be completely clear this is not an issue per se but rather a clarification request as the title states. What does the "transform" parameter mean in the context of the `microscope.Camera` base class?

I'm currently implementing microscope on an external software (ImSwitch, I've been mentioning this in the past already) and it's going quite smoothly. What I don't understand is how to use this setting, what it implies and so on. I tried looking into the documentation and in the repo for more clarification about it, but I only managed to recover an issue concerning the fact that some methods related to this "transform" parameter have been removed.

I apologize if there was a doc page I missed somehow!

iandobbie, 2023-02-14:
I suspect it is not fully documented. 

The transform is a three component tuple, with each component a boolean. These define 8 possible states which are all possible 90 degree rotations and mirror flips to get a 2d image to the orientation required. The bits actually represent vertical flip, horizontal flip and a single 90 degree rotation. 

This allows easy computational operations to provide images which properly align with the stage motion and situations like multiple cameras separated by a dichroic (so one if mirrored compared to the other) to align, or images from an EMCCD in EM mode and conventional CCD mode to be the same orientation (these two modes read the chip out from different sides and produce images at different orientations). 

I find it very difficult to work out the operations needed to get the orientation required when setting up a new system so I usually just use a mosaic tool and fiddle till it moves the way I expect and the image edges align correctly. There aren't that many possible states so trial and error is good enough.

This may seem like a complex way to represent this function but it is defined this way for its easy computation.

jacopoabramo, 2023-02-14:
Thanks @iandobbie , I suspected it was something related to this but wanted to be sure.
And this is transparent to the camera, correct? Meaning that the same flip can occur with different camera models and without affecting the interaction with the specific API set?

iandobbie, 2023-02-14:
It should be completely transparent to the camera. However, a specific implementation might choose to implement it either use the python code in the abc, or override that and utilise a call to the low level camera api to implement the same functionality.

jacopoabramo, 2023-02-14:
Perfect, thank you. Closing the issue.

closed 2023-02-14 by jacopoabramo