python-microscope/microscope #278: Question: troubles by loading the camera modules pvcam and andorsdk3

PiaPritzke, 2023-08-14:
Dear all,

I have a question regarding an issue I encountered while trying to load the camera modules pvcam and andorsdk3. I receive the following error message

For pvcam:
File ~\AppData\Local\Programs\Python\Python39\lib\importlib\__init__.py:127, in import_module(name='microscope.cameras.pvcam', package=None)
    125             break
    126         level += 1
--> 127 return _bootstrap._gcd_import(name[level:], package, level)
        level = 0
        name = 'microscope.cameras.pvcam'
        name[level:] = 'microscope.cameras.pvcam'
        package = None
        _bootstrap = <module 'importlib._bootstrap' (frozen)>

File <frozen importlib._bootstrap>:1030, in _gcd_import(name='microscope.cameras.pvcam', package=None, level=0)

File <frozen importlib._bootstrap>:1007, in _find_and_load(name='microscope.cameras.pvcam', import_=<function _gcd_import>)

File <frozen importlib._bootstrap>:986, in _find_and_load_unlocked(name='microscope.cameras.pvcam', import_=<function _gcd_import>)

File <frozen importlib._bootstrap>:680, in _load_unlocked(spec=ModuleSpec(name='microscope.cameras.pvcam', load...b\\site-packages\\microscope\\cameras\\pvcam.py'))

File <frozen importlib._bootstrap_external>:850, in exec_module(self=<_frozen_importlib_external.SourceFileLoader object>, module=<module 'microscope.cameras.pvcam' from 'C:\\Use...b\\site-packages\\microscope\\cameras\\pvcam.py'>)
File <frozen importlib._bootstrap>:228, in _call_with_frames_removed(f=<built-in function exec>, *args=(<code object <module> at 0x00000186DE0C3EA0, fil...e-packages\microscope\cameras\pvcam.py", line 24>, {'ACC_EXIST_CHECK_ONLY': 
3, 'ACC_READ_ONLY': 1, 'ACC_READ_WRITE': 2, 'ACC_WRITE_ONLY': 4, 'ACQUISITION_IN_PROGRESS': 5, 'ATTR_ACCESS': 7, 'ATTR_AVAIL': 8, 'ATTR_COUNT': 1, 'ATTR_CURRENT': 0, 'ATTR_DEFAULT': 5, ...}), **kwds={})

File ~\AppData\Local\Programs\Python\Python39\lib\site-packages\microscope\cameras\pvcam.py:681
    679         _lib = ctypes.WinDLL("pvcam32")
    680     else:
--> 681         _lib = ctypes.WinDLL("pvcam64")
    682 else:
    683     _lib = ctypes.CDLL("pvcam.so")

File ~\AppData\Local\Programs\Python\Python39\lib\ctypes\__init__.py:374, in CDLL.__init__(self=<WinDLL 'pvcam64', handle 0>, name='pvcam64', mode=4096, handle=None, use_errno=False, use_last_error=False, winmode=None)
    371 self._FuncPtr = _FuncPtr
    373 if handle is None:
--> 374     self._handle = _dlopen(self._name, mode)
        self._name = 'pvcam64'
        self = <WinDLL 'pvcam64', handle 0 at 0x186de0c1af0>
        self._handle = 0
        mode = 4096
        _dlopen = <built-in function LoadLibrary>
    375 else:
    376     self._handle = handle

FileNotFoundError: Could not find module 'pvcam64' (or one of its dependencies). Try using the full path with constructor syntax.

and for andorsdk3:
File ~\AppData\Local\Programs\Python\Python39\lib\site-packages\microscope\cameras\andorsdk3.py:34
     32 import microscope
     33 import microscope.abc
---> 34 from microscope.cameras import _SDK3 as SDK3
     35 from microscope.cameras._SDK3Cam import (
     36     ATBool,
     37     ATCommand,
   (...)
     42     ATString,
     43 )
     46 _logger = logging.getLogger(__name__)

File ~\AppData\Local\Programs\Python\Python39\lib\site-packages\microscope\cameras\_SDK3.py:37
     34 _stdcall_libraries = {}
     36 if os.name in ("nt", "ce"):
---> 37     _stdcall_libraries["ATCORE"] = ctypes.WinDLL("atcore")
        _stdcall_libraries = {}
     38     _stdcall_libraries["ATUTIL"] = ctypes.WinDLL("atutility")
     39     CALLBACKTYPE = ctypes.WINFUNCTYPE(c_int, AT_H, POINTER(AT_WC), c_void_p)

File ~\AppData\Local\Programs\Python\Python39\lib\ctypes\__init__.py:374, in CDLL.__init__(self=<WinDLL 'atcore', handle 0>, name='atcore', mode=4096, handle=None, use_errno=False, use_last_error=False, winmode=None)
    371 self._FuncPtr = _FuncPtr
    373 if handle is None:
--> 374     self._handle = _dlopen(self._name, mode)
        self._name = 'atcore'
        self = <WinDLL 'atcore', handle 0 at 0x186dc9abeb0>
        self._handle = 0
        mode = 4096
        _dlopen = <built-in function LoadLibrary>
    375 else:
    376     self._handle = handle

FileNotFoundError: Could not find module 'atcore' (or one of its dependencies). Try using the full path with constructor syntax.

I’m not sure where this problem comes from or how to solve it, so any help or comments would be much appreciated!

carandraug, 2023-08-22:
Assuming you do have pvcam and andor sdk3 installed, and since you're using Python 3.9, this is likely a duplicate of #235 . While we get around to fix this, this can be fixed by copying the pvcam and andor dll into `~\AppData\Local\Programs\Python\Python39\lib\site-packages\microscope\cameras\`. Closing as duplicated.

closed 2023-08-22 by carandraug

PiaPritzke, 2023-09-19:
Dear @iandobbie  and @carandraug,
Thank you very much for your help. I followed your advice and copied the dlls to ~\AppData\Local\Programmes\Python\Python39\lib\site-packages\microscope\cameras\. This worked for the pvcam, but I still get the same error message as before from the andorsdk3.

PiaPritzke, 2023-09-21:
Dear @carandraug,

Thank you for your answer.
I already copied all the dll files in that directory, from the pvcam and the andor. I also checked that atcore.dll and atutility.dll are in the directory.
If you have any other suggestion or idea what I could try next, I would be very grateful.

juliomateoslangerak, 2023-09-21:
Hi,
Here is a screenshot of my folder with a working Zyla4.2. I'm not sure which dll's are necessary or depending on each other. At the time I probably copied everything.
[image]

iandobbie, 2023-09-21:
Thanks Julio, that is a useful reference point.

@PiaPritzke do you have all those dll's starting the 'at' in the screen shot above? If so can you run it again and paste in the error messages again, even if the error seems the same it may be another dll it is failing on. 

Sorry you are having trouble with this, we have had similar and very frustrating issues with the Andor dll before. I our case it was loading older versions we didn't realise we had, so maybe also search for other copies of the atcore.dll and see if they might be accessed by accident. Maybe move them off to a new directory temporarily and see if the andor software still works and if it changes anything in microscope.