python-microscope/microscope #282: simulated camera issue with font generation.

iandobbie, 2023-09-12:
I recently moved to python 3.11.5 and something has gone wonky on my end. Simulated cameras dont return any data and produce the error. 

"""
2023-09-12 17:56:48,402:TestCamera (microscope.abc):ERROR:PID 16353: in _fetch_loop:
Traceback (most recent call last):
  File "/Users/ID/src/microscope/microscope/abc.py", line 726, in _fetch_loop
    data = self._fetch_data()
           ^^^^^^^^^^^^^^^^^^
  File "/Users/ID/src/microscope/microscope/simulators/__init__.py", line 251, in _fetch_data
    image = self._image_generator.get_image(
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/ID/src/microscope/microscope/simulators/__init__.py", line 108, in get_image
    size = tuple(d + 2 for d in self._font.getsize(text))
                                ^^^^^^^^^^^^^^^^^^
AttributeError: 'ImageFont' object has no attribute 'getsize'
"""
I assume I have a wrong package version or the API has changed recently and I have a newer than expected version.

iandobbie, 2023-09-12:
I think the API has moved. I am now on PIL 10.0.0 and it appears I need either
ImageFont.getbbox() or ImageFont.getlength()

iandobbie, 2023-09-12:
A little investigation of the code, its needs both the X and y sizes of the text, so I suggest we replace the original:

            size = tuple(d + 2 for d in self._font.getsize(text))

With a slight modification:
            size = tuple(d + 2 for d in self._font.getbbox(text)[2:])

this is around line 108 of microscope/simulators.__init__.py

PIL apparently introduced getbbox in version 8.0.0 can just do this or do we need to check version and change the call? Or maybe just update the requirements to PIL> 8.0.0

carandraug, 2023-09-12:
PIL 8.0.0 added `FreetypeFont.getbbox` but `ImageFont.getbbox` that we need was only added in version 9.2 and that was only released July 2022. If we can support both that would be nicer.

carandraug, 2023-09-12:
Now fixed in 410b472 by checking if `getbbox` is available and using `getsize` only if not. Closing as fixed.

closed 2023-09-12 by carandraug

iandobbie, 2023-09-12:
Ok I’ll check versions and call the relevant function based on that. I wonder when getsize went away?

Ian