python-microscope/microscope #283: ASI controller improvements required.

iandobbie, 2023-09-21:
With help from Jon from ASI (thank you!) I finally managed to sort out my annoying issue with the ASI stage so it is now merged. I think there are a few issues. 

1) Documentation - needs an update for this controller.

2) Code currently has arbitrary values for spin rates to reach hard stops and then for programmed stage speeds. This need to be more sensibly set, but also added as config parameters so they can be set in config files for specific systems.

3) Checking to ensure it reflects the real hardware. This is very much a clone of the ludl controller as the command set is based on the ludl command set. However I have not made any effort to ensure anything beyond the commands required to get a basic xy stage are working. 

I suspect there will be others.

iandobbie, 2023-09-22:
For point 2, Jon from ASI recommended that we set the speed very high, then query the stage speed as the firmware limits it to what the hardware can actually achieve. We then obviously need to backoff some as at this speed the Y axis fails on my stage. The manual then says "Default speed is ~67% of the max speed." 

I will add code to do this. We could also move away from the spin to reach the end stop but this seems to work well and obviously the position tracking keeps going as it produces realistic values for the total stage range.

iandobbie, 2023-09-22:
Added notes to the NEWS and supported devices in [751e37f](https://github.com/python-microscope/microscope/commit/751e37f7120e9a24b8c6a2569b43ceb1e332f00b)

carandraug, 2023-09-25:
If the Ludl and ASI stages have the same API can we have them share their code instead of copying the code and diverging?

iandobbie, 2023-10-05:
No they don't have the same API. The ASI command set was originally  based on the ludl controller but there are real differences. I think trying to have one module which then switches modes would be more confusing than just having separate modules which are similar but not identical. 

Two issue that immediately spring to mind is the two systems treat line endings differently and that the spin speed mechanism I used fine on the ludl just didn't work on the ASI controller.

juliomateoslangerak, 2023-10-23:
@iandobbie You mentioned some speed limitations in the comm with the stage. Looking into the docs it seems that the baudrate is configurable with the dip switches up to 115200 [ref](https://asiimaging.com/docs/tech_note_rs232_comm)

iandobbie, 2023-10-27:
The speed I am talking about is the motor speeds, that I have just pushed a fix for. As suggested in the manual I set the speed to an unreasonably high value, then read back the speed set by the controller and finally set the speed to 67% of this. I have pushed this into master as it worked for me and is the documented method to set a sensible speed on the motors.

As to the final issue of true compatibility for other devices, I have no access to devices other than the stage I have so I will close this issue and open future issues with more details if we find other devices on this controller need more work.

closed 2023-10-27 by iandobbie