python-microscope/microscope #290: Pillow 10 removes 'getsize' from 'FreeTypeFont'

juliomateoslangerak, 2023-10-30:
Getting images from the simulatedCamera returns this exception: ```'ImageFont' object has no attribute 'getsize'```

It seems that getsize was removed from the FreeTypeFont. (https://github.com/tensorflow/models/issues/11040)

The fix seems to be using ```getbbox```. I tested replacing in the simulated camera getImage method:
```
            size = tuple(d + 2 for d in self._font.getsize(text))
```
with 
```
            size = tuple(d + 2 for d in self._font.getbbox(text)[-2:])
 ```

Con you guys validate this?

carandraug, 2023-10-30:
This was reported in #282 and should have been fixed in [410b472](https://github.com/python-microscope/microscope/commit/410b47215d2037b90a2c8bd92d3fd7ab6ce29bb8). What you using development sources?

juliomateoslangerak, 2023-10-30:
Oops! Sorry. 

I'm using the development source but I did not pull since I did the installation a few weeks ago.
Closing

closed 2023-10-30 by juliomateoslangerak