python-microscope/microscope #291: Hamamatsu camera symbol lookup error

juliomateoslangerak, 2023-10-30:
I'm trying to connect with a Hamamatsu camera (ORCA-Spark) in Linux (Ubuntu 23.04) and I'm getting the following error when I try to get the list of cameras.
```bash
(venv) julio@img-mifobio:~/PycharmProjects/microscope$ python microscope/cameras/hamamatsu.py list-devices
python: symbol lookup error: /usr/local/hamamatsu_dcam/api/modules/libdcamdig.so: undefined symbol: _Z12get_filepathPciS_
```
same error calling _list_devices directly or trying to initialize a camera with some id (0, "0", 1, "1").
I traced the error to line 230 (_DCAM_API.__init__)
``` python
                status = dcam.api_init(ctypes.byref(_DCAM_API._api))
```
I checked with Hamamatsu's own tool and it reports properly a camera connected:
```bash
julio@img-mifobio:~/Apps/DCAM-API_for_Linux_v4.0.6644/tools/x86_64$ ./dcamcfgc 
dcamcfgc: DCAM Configurator ver 23.6.6644
Initializing DCAM-API...
Found a camera
1:	C11440-36U (S/N: 000424)
[Parameter List]
--- C11440-36U (S/N: 000424) Ver:1.12.C ---
None
hit the Enter key to finish!
```

Any idea of can be wrong?

iandobbie, 2023-10-30:
My first thought is are you definitely using the same dll? I remember that Mick had a nightmare with the Andor cameras and eventually found that he had two different sets of dll's in different directories that were being accessed by the the vendor software and the python code.

juliomateoslangerak, 2023-10-31:
A system wide search shows only one SO files and the other symlinks to that file, which is actually the one pointed by the error:

/usr/local/hamamatsu_dcam/api/modules/libdcamdig.so.4.0.6644

If I move the Hamamatsu tool to another place it still finds the camera, which I assume means that the SO is in the right place.
```_LIB = microscope._utils.library_loader("libdcamapi.so", ctypes.CDLL)``` also works.

I did some raw tests to find the function and it finds the pointer to the function
```python
>>> from ctypes import cdll, CDLL
>>> cdll.LoadLibrary("libdcamapi.so")
<CDLL 'libdcamapi.so', handle 1686b40 at 0x7fc806599710>
>>> libc = CDLL("libdcamapi.so")
>>> libc._Z12get_filepathPciS_
<_FuncPtr object at 0x7fc80641c7a0>
```

carandraug, 2023-11-01:
The error you get is complaining about a missing symbol in `libdcamdig.so` which is not the one we load in Microscope. We load `libdcamapi` and I'm assuming that `libdcamdig` is load dynamically by it.

You seem to be using DCAM-Api version "4.0.6644" while I was using version "4.0.5868" when developing this. I looked at the version I have and seems to not have that symbol. Maybe I send it to you?

juliomateoslangerak, 2023-11-02:
Indeed that seems to be like that.
```
julio@img-mifobio:/usr/local$ ll hamamatsu_dcam/api/
total 216
drwxr-xr-x 4 root root   4096 Oct 31 16:24 ./
drwxr-xr-x 4 root root   4096 Oct 31 16:24 ../
drwxr-xr-x 3 root root   4096 Oct 31 16:24 etc/
lrwxrwxrwx 1 root root     22 Oct 31 16:24 libdcamapi.so.4 -> libdcamapi.so.4.0.6644*
-rwxr-xr-x 1 root root 200504 Oct 31 16:24 libdcamapi.so.4.0.6644*
drwxr-xr-x 2 root root   4096 Oct 31 16:24 modules/
-rwxrwxrwx 1 root root   1805 Oct 31 16:24 uninstall_runtime.sh*
julio@img-mifobio:/usr/local$ ll hamamatsu_dcam/api/modules/
total 6508
drwxr-xr-x 2 root root    4096 Oct 31 16:24 ./
drwxr-xr-x 4 root root    4096 Oct 31 16:24 ../
lrwxrwxrwx 1 root root      22 Oct 31 16:24 libdcamdig.so -> libdcamdig.so.4.0.6644*
lrwxrwxrwx 1 root root      22 Oct 31 16:24 libdcamdig.so.4 -> libdcamdig.so.4.0.6644*
-rwxr-xr-x 1 root root 5840544 Oct 31 16:24 libdcamdig.so.4.0.6644*
lrwxrwxrwx 1 root root      21 Oct 31 16:24 libfgusb3.so -> libfgusb3.so.4.0.6644*
lrwxrwxrwx 1 root root      21 Oct 31 16:24 libfgusb3.so.4 -> libfgusb3.so.4.0.6644*
-rwxr-xr-x 1 root root  814528 Oct 31 16:24 libfgusb3.so.4.0.6644*
```