python-microscope/microscope #294: Camera pixel positions should always start from 0

iandobbie, 2023-11-01:
A recent discussion brought up that some cameras start counting pixels at 0,0 (eg ximea cameras) and some start from pixel 1,1 (eg Andor cameras). 

I propose we should specify that camera pixel positions eg for ROI definitions, should always start at 0,0. This is expected behavior in Python. If the lower level interface expects counting to start at 1,1 then the hardware specific microscope code should deal with the translation of this to 0,0.

juliomateoslangerak, 2023-11-03:
Looking at abc camera code, it seems that the ROI is specified as:
- left and top values are specified in sensor-pixels
- width and height is in binned pixels (size of the image data received)

So ```roi.left + (roi.width * binning.h) =< sensor.width```. This logic is implemented as a validation in some cameras. We could move it to abc.

Is this how it is supposed to be? If so:
- Ximea camera is doing the right thing
- Andor camera just has correct the indexing of the left and top (1 based)
- Hamamatsu camera (working on this now) provides width and height in sensor pixels so a correction has to be applied

juliomateoslangerak, 2023-11-03:
I pushed some of the changes discussed here to this branch. Can you have a look?