python-microscope/microscope #299: more improvements to ASi stage

carandraug, 2024-01-09:
@joris-afk and @aasmaa1 (from Mike Strauss) have made improvements to the ASI stage. Not sure what but they are tangled with initial support for PCO cameras (see #118).  Merging this work will likely cause conflicts since they diverged before the ASi improvements done in #283 .

iandobbie, 2024-01-11:
I have had a quick look and their code is an edit of the stage/asi code that I originally checked in but moved to controller and expanded to cope with more than stages. I think it will take some  time to work through the changes and decide on the best way to merge the two branches. 

I will start trying to decide how best to merge. They also have a config file which should allow us to understand exactly how they were using the setup. 

I think I will separate the asi changes from the pco ones and try to work through the asi changes.

iandobbie, 2024-01-12:
I have had more of a look through the repository and there are quite a few wrinkles. 

They seem to have merged and then deleted a whole load of changes from the main microscope repository so we have a few commits which change like 50 files. 

They have added a couple of directories at the root level of the repository. 

I think the only way to integrate any of this into the main repository with a reasonable amount of work is to just pull the final pco.py an asi.py files and look at them, I dont think the hsitory has much useful going forward.

iandobbie, 2024-01-12:
Ok a quick comparison between the asi controller code from me and Julio and the stuff from McGill shows there are totally separate code bases with no common code. 

Our code is more in the microscope style, however their code has some functionality which our lacks, like setting velocity and acceleration etc...

I need to go through the code and compare features. Then copy the missing and useful features from their code into the asi.py in the controllers directory.

iandobbie, 2024-01-12:
List of missing functionality in controllers implementation:

1) axis acceleration
2) axis settle time
3) axis screw pitches and matching to resolution and speed
4) axis min and max limits from config, still probably needs homing on initialization for absolute positioning.
5) LEDs, seem to have 3 ttl modes - on, off and pwm, need to check manual.
6) get and set pwm intensity

carandraug, 2024-01-12:
When copying things over, don't forget their tests. They wrote a lot of testing code.

iandobbie, 2024-01-12:
Yes, but probably needs to be rewritten to match up with the different code if we are using the base code inthe controller directory. 

Should this test code go into testsuite, or is that just for simulated devices?

carandraug, 2024-01-12:
> Should this test code go into testsuite, or is that just for simulated devices?

I would argue for a new module under `microscope.testsuite`

iandobbie, 2024-01-12:
Ok, I will start working on that. Not looked at the test code at all yet.