python-microscope/microscope #305: Controlling Nikon Eclipse Ti Microscope Focus with python-microscope

shay82, 2024-08-08:
Hello,

I have a _Nikon Eclipse Ti microscope_ with the relevant drivers installed. 
Using the Nikon _Ti control tool_, I can control the focus manually.

I am interested in controlling the focus using the python-microscope package. Is this currently possible with this package? If so, could you please provide an example of how to achieve this?

Additionally, are there any auto-focus tools available within the python-microscope package that I could use?

Thank you for your assistance!

carandraug, 2024-08-08:
> I am interested in controlling the focus using the python-microscope package. Is this currently possible with this package? If so, could you please provide an example of how to achieve this?

There is currently no support for the Nikon Eclipse Ti microscope.  You are welcome to add support to it, see our guidelines to add support for new devices at https://python-microscope.org/doc/get-involved/new-device (you will need the Nikon Ti SDK).

> Additionally, are there any auto-focus tools available within the python-microscope package that I could use?

None of the stages currently supported by Python-Microscope have auto-focus.

iandobbie, 2024-08-08:
> > Additionally, are there any auto-focus tools available within the python-microscope package that I could use?
> 
> None of the stages currently supported by Python-Microscope have auto-focus.

I think the question is about image based auto focus within microscope. This has not been implemented but would be relatively easy with a controller that wraps the light source, camera and stage. The question is what metric you use to determin the focal plane.

iandobbie, 2024-08-08:
> I have a _Nikon Eclipse Ti microscope_ with the relevant drivers installed. Using the Nikon _Ti control tool_, I can control the focus manually.

I can't remember the exact situation with Nikon but our experience with the motorized stands is that the manufactures require an MTA that restricts redistribution of the code making open source release of a controller impossible without a closed source binary blob. Fundamentally I disapprove of this approach and I feel that the stand manufactures are missing out as if one stand was easily supported by open source platforms it could significantly add to their sales. I have had this discussion will all the manufactures many times to no avail.

iandobbie, 2024-08-08:
> I can't remember the exact situation with Nikon but our experience with the motorized stands is that the manufactures require an MTA that restricts redistribution of the code making open source [...]

This is the fact for Nikon see the micromanager support at https://micro-manager.org/NikonTI The license is 

License: | Source code cannot be made available

shay82, 2024-08-08:
> There is currently no support for the Nikon Eclipse Ti microscope. You are welcome to add support to it, see our guidelines to add support for new devices at [...]

Thank for your response.

shay82, 2024-08-08:
> I can't remember the exact situation with Nikon but our experience with the motorized stands is that the manufactures require an MTA that restricts redistribution of the code making open source [...]

I agree.
Thank you